pguyot/arm-runner-action #3: Copy artifact inside image to outside

patrickelectric, 2021-06-22:
Is that possible ? 
I did try to use the parameters `copy_artifact_path`, but it appears something outside the image to be inside, but I'm looking the other way around to output the binary.

pguyot, 2021-06-22:
With `copy_artifact_path` and `copy_artifact_dest`, you can copy an artifact from the image (generated by your build commands), outside the image, before the image is unmounted.

The actual copy command is:
```
sudo cp -Rp ${{ steps.mount_image.outputs.mount }}${repository_path}/${{ inputs.copy_artifact_path }} ${{ inputs.copy_artifact_dest }}
```
https://github.com/pguyot/arm-runner-action/blob/main/action.yml#L96

The usage would be something like:
```
jobs:
  build:
    runs-on: ubuntu-latest
    steps:
    - uses: actions/checkout@v2
    - uses: pguyot/arm-runner-action@v1
      id: build_image
      copy_artifact_path: /tmp/my-file.txt
      copy_artifact_dest: .
      with:
        commands: |
            touch /tmp/my-file.txt
    - name: Upload artifact
      uses: actions/upload-artifact@v2
      if: github.ref == 'refs/heads/releng' || startsWith(github.ref, 'refs/tags/')
      with:
        name: Generated artifact
        path: my-file.txt
```

Is it what you are trying to achieve?

patrickelectric, 2021-06-22:
I just tested and it does not work, you can check it running here:
https://github.com/patrickelectric/mavlink-router/runs/2889100688?check_suite_focus=true#step:4:1

```
arm-v6:
  name: ARM v6 binary
  runs-on: ubuntu-latest
  steps:
    - uses: actions/checkout@v2
      with:
        submodules: recursive
    - uses: pguyot/arm-runner-action@v1
      with:
        copy_artifact_path: touch mavlink-routerd-libc-armv6
        image_additional_mb: 1024
        commands: |
          touch mavlink-routerd-libc-armv6
    - name: Rename
      run: ls
    - uses: actions/upload-artifact@master
      with:
        path: mavlink-routerd-libc-armv6
```

pguyot, 2021-06-24:
There is a typo in your yml (`copy_artifact_path: touch mavlink-routerd-libc-armv6` => `copy_artifact_path: mavlink-routerd-libc-armv6`), but this doesn't explain the bug which I could reproduce.

patrickelectric, 2021-06-24:
Oh, thanks! Sorry that. Let me know if I can be of any help.

pguyot, 2021-06-25:
The bug is caused by default being effectless when accessing input values with this syntax:
```
${{ inputs.copy_artifact_dest }}
```

As a workaround, you can define `copy_artifact_dest` as suggested above and implemented here:
https://github.com/pguyot/mavlink-router/commit/d3e701ade1dcf3d312c29509d85b9f8a7d8dd159

I am keeping this bug open to test and fix every use of parameters that have default values.

pguyot, 2021-06-27:
In fact, the bug is caused by a typo. The proper key for default values is `default` and not `defaut`.
Found this out when fixing this bug and writing tests for it.

f-laurens, 2021-06-27:
> In fact, the bug is caused by a typo. The proper key for default values is `default` and not `defaut`.
Arrgh... was too obvious to spot ! ;-) Looks like the erroneous `defaut: ` line was copy/pasted a number of times...

commit c88b1c2f (2021-06-27): Add tests as workflows for this action
Also fix #3 which was caused by a typo
commit d59f55ef (2021-06-27): Add tests as workflows for this action
Also fix #3 which was caused by a typo
commit a80e9ba7 (2021-06-27): Add tests as workflows for this action
Also fix #3 which was caused by a typo
PR #6 (2021-06-27): Add tests as workflows for this action
Also fix #3 which was caused by a typo
closed 2021-06-27 by pguyot