pguyot/arm-runner-action #12: Improvement: add the possibility to use any image from url or from path

2010019970909, 2021-08-22:
Hi,

I'm currently trying to build userland (to build raspimjpeg) on the arm-runner-action, but cmake has a bug in the latest available version in Raspberry Pi OS, so I must download cmake from source and build it before building using cmake to build userland... and it takes around 3h 40min. So being able to have a cached image would be a great addition...

I do not have the knowledge to add this feature, but I suppose that the url or the path to the image could be passed using the base_image input.

Regards,
Vincent

P.S.: you can close this issue if it is not relevant.

pguyot, 2021-08-22:
Makes sense. Alternatively, you can always install binaries such as a dpkg that would be saved elsewhere.

twojstaryzdomu, 2021-08-24:
PR #13 adds support for custom urls as base_image.

pguyot, 2022-02-13:
Fixed in #13 / #20

closed 2022-02-13 by pguyot