pguyot/arm-runner-action #15: problems on private runner

Apfelwurm, 2021-09-08:
hi together, i've setted up a self hosted runner on an ubuntu 18.04 LTS azure machine because on the default runners my build exceeds the 6 hour job limit.
I got one working run on the self hosted runner and now it seems to don't work anymore.
Therefore i simplified the action to the minimum so only a echo is executed: https://github.com/Apfelwurm/cmake-raspberry/blob/main/.github/workflows/build_release.yml

Even with this minimal setup it results in the following behavior:

the action runs and executes ``sudo chroot /home/actions/temp/arm-runner/mnt /tmp/commands.sh``, at this point qemu-arm-static begins to use between 99.3% and 100% cpu (top output) but nothing more happens.
If i wait around 15 minutes, chroot exits with code 255. So its possible that i encounter the same problem that is mentioned [here](https://github.com/pguyot/arm-runner-action/issues/5) but in a different situation. Sadly i only have that one self hosted runner, therefore different jobs would run on the same machine.

let me know when there is more information needed and how i can get it.

Ty!

pguyot, 2022-02-13:
The link seems to be dead. Were you able to solve the problem?

Apfelwurm, 2022-02-24:
Hey, sorry for the dead link, was not that genius to send the link of a changing file :D 
https://github.com/Apfelwurm/cmake-raspberry/blob/0951880bc9f4fb736885ea885b44d79f70606805/.github/workflows/build_release.yml

I have not tried it since september because i could reduce the build time so that it works again on githubs runners. If you want, i can test this again in the next 1-2 Weeks.

pguyot, 2022-03-03:
Could you please test, ideally with `uses: pguyot/arm-runner-action@main` ?
Several bugs were fixed recently.

pguyot, 2022-04-02:
Closing this, please feel free to reopen if the problem still exists.

closed 2022-04-02 by pguyot