pguyot/arm-runner-action #26: Issue with CPU cortex-a8 and raspi_4_bullseye:20220121

thefastman, 2022-02-21:
Hello! 
I'm trying to get work runner for RPI4 with 64bit OS

My workflow is:

```yaml
steps:
    - uses: actions/checkout@v2
    - uses: pguyot/arm-runner-action@HEAD
      id: build_image
      with:
        cpu: cortex-a8
        base_image: raspi_4_bullseye:20220121
        copy_repository_path: /home/pi/app
        image_additional_mb: 2048
        commands: |
            sudo cat /etc/os-release
            sudo lscpu
```

And it fails with error:
```2022-02-21T11:22:18.7910107Z qemu-aarch64-static: unable to find CPU model 'cortex-a8'```

What did I do wrong?
Thanks.

pguyot, 2022-02-21:
Please try with `cpu: cortex-a53`. `cortex-a8` is the 32 bits version, and you picked a 64 bits image (raspi_4_bullseye:20220121).

thefastman, 2022-02-22:
Works. Thanks !

closed 2022-02-23 by pguyot