pguyot/arm-runner-action #38: Is there a posibility to include Nvidia Jetson (ARM64)

raphaelscholle, 2022-05-31:
Hi, I just started using this awesome project and replace our self-hosted runners,
since you're basically emulating an ARM-cpu and download raspberry images, is there a possibility to also include Jetson images ?

pguyot, 2022-05-31:
It should be doable, especially if they are really based on ubuntu linux.

This action considers partition 1 is boot and 2 is root, and the glob expression doesn't work with more than 9 partitions.
Apparently, the Jetson image I tried has 11 partitions. Do you know how they are organized and how they should be mounted?

https://github.com/pguyot/arm-runner-action/runs/6679373325?check_suite_focus=true#step:3:157

raphaelscholle, 2022-05-31:
[https://geekymind.de/jetson.png](url)

looks like mmcblk0p1 is the main partition, the rest is just for the bootloader

pguyot, 2022-06-01:
Thanks. Got it working. Fixed in #39
Can you test it before we can call it a release?

The CI test is here:
https://github.com/pguyot/arm-runner-action/blob/main/.github/workflows/test-partitions.yml

raphaelscholle, 2022-06-02:
mounts and starts great, but I have a strange issue, networking isn't working (using jetson-nano-image) I'll invenstigate, maybe we only need dhclient or something like that :

> 2022-06-02T22:43:28.1450723Z ##[group]Run pguyot/arm-runner-action@f204b58258d5b01e297eec6db036a14b11dc9ffa
2022-06-02T22:43:28.1451317Z with:
2022-06-02T22:43:28.1451991Z   base_image: https://developer.download.nvidia.com/embedded/L4T/r32_Release_v6.1/Jetson_Nano_2GB/jetson-nano-2gb-jp46-sd-card-image.zip
2022-06-02T22:43:28.1452697Z   cpu: cortex-a53
2022-06-02T22:43:28.1453038Z   rootpartition: 1
2022-06-02T22:43:28.1453524Z   commands: cat /etc/os-release
uname -a
ping google.de
sudo apt update
sudo apt upgrade

2022-06-02T22:43:28.1454047Z   image_additional_mb: 0
2022-06-02T22:43:28.1454421Z   bind_mount_repository: no
2022-06-02T22:43:28.1454766Z   copy_artifact_dest: .
2022-06-02T22:43:28.1455108Z   optimize_image: yes
2022-06-02T22:43:28.1455470Z   use_systemd_nspawn: no
2022-06-02T22:43:28.1455786Z   shell: /bin/sh
2022-06-02T22:43:28.1456104Z   exit_on_fail: yes
2022-06-02T22:43:28.1456424Z   debug: yes
2022-06-02T22:43:28.1456759Z   import_github_env: no
2022-06-02T22:43:28.1457085Z   export_github_env: no
2022-06-02T22:43:28.1457441Z env:
2022-06-02T22:43:28.1457743Z   BUILD_TYPE: Release
2022-06-02T22:43:28.1458071Z ##[endgroup]
2022-06-02T22:43:28.1713819Z ##[group]Run sudo apt-get update && sudo apt-get install -y qemu qemu-user-static binfmt-support parted wget dosfstools zip
2022-06-02T22:43:28.1714708Z [36;1msudo apt-get update && sudo apt-get install -y qemu qemu-user-static binfmt-support parted wget dosfstools zip[0m
2022-06-02T22:43:28.1780681Z shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
2022-06-02T22:43:28.1781168Z env:
2022-06-02T22:43:28.1781463Z   BUILD_TYPE: Release
2022-06-02T22:43:28.1781788Z ##[endgroup]
2022-06-02T22:43:28.3375105Z Hit:1 http://azure.archive.ubuntu.com/ubuntu focal InRelease
2022-06-02T22:43:28.3377574Z Get:2 http://azure.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
2022-06-02T22:43:28.3386334Z Get:3 http://azure.archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
2022-06-02T22:43:28.3410425Z Get:4 http://azure.archive.ubuntu.com/ubuntu focal-security InRelease [114 kB]
2022-06-02T22:43:28.3929991Z Get:5 https://packages.microsoft.com/ubuntu/20.04/prod focal InRelease [10.5 kB]
2022-06-02T22:43:28.5879063Z Hit:6 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu focal InRelease
2022-06-02T22:43:28.6065525Z Get:7 http://azure.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [1833 kB]
2022-06-02T22:43:28.6391408Z Get:8 http://azure.archive.ubuntu.com/ubuntu focal-updates/main Translation-en [337 kB]
2022-06-02T22:43:28.6426357Z Get:9 http://azure.archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [15.4 kB]
2022-06-02T22:43:28.6463785Z Get:10 http://azure.archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1010 kB]
2022-06-02T22:43:28.6616828Z Get:11 http://azure.archive.ubuntu.com/ubuntu focal-updates/restricted Translation-en [144 kB]
2022-06-02T22:43:28.6782987Z Get:12 http://azure.archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [926 kB]
2022-06-02T22:43:28.6843543Z Get:13 http://azure.archive.ubuntu.com/ubuntu focal-updates/universe Translation-en [208 kB]
2022-06-02T22:43:28.6895651Z Get:14 http://azure.archive.ubuntu.com/ubuntu focal-updates/universe amd64 c-n-f Metadata [20.7 kB]
2022-06-02T22:43:28.7635820Z Get:15 http://azure.archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [23.7 kB]
2022-06-02T22:43:28.7652739Z Get:16 http://azure.archive.ubuntu.com/ubuntu focal-backports/universe amd64 c-n-f Metadata [860 B]
2022-06-02T22:43:28.9253114Z Get:17 http://azure.archive.ubuntu.com/ubuntu focal-security/main amd64 Packages [1491 kB]
2022-06-02T22:43:28.9514668Z Get:18 http://azure.archive.ubuntu.com/ubuntu focal-security/main Translation-en [257 kB]
2022-06-02T22:43:28.9519646Z Get:19 http://azure.archive.ubuntu.com/ubuntu focal-security/main amd64 c-n-f Metadata [10.4 kB]
2022-06-02T22:43:28.9545951Z Get:20 http://azure.archive.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [948 kB]
2022-06-02T22:43:28.9675020Z Get:21 http://azure.archive.ubuntu.com/ubuntu focal-security/restricted Translation-en [135 kB]
2022-06-02T22:43:28.9742791Z Get:22 http://azure.archive.ubuntu.com/ubuntu focal-security/universe amd64 Packages [705 kB]
2022-06-02T22:43:28.9867853Z Get:23 http://azure.archive.ubuntu.com/ubuntu focal-security/universe Translation-en [126 kB]
2022-06-02T22:43:28.9879888Z Get:24 http://azure.archive.ubuntu.com/ubuntu focal-security/universe amd64 c-n-f Metadata [14.5 kB]
2022-06-02T22:43:29.0622012Z Get:25 https://packages.microsoft.com/ubuntu/20.04/prod focal/main amd64 Packages [161 kB]
2022-06-02T22:43:35.4390452Z Fetched 8711 kB in 2s (4549 kB/s)
2022-06-02T22:43:36.5074377Z Reading package lists...
2022-06-02T22:43:36.6064760Z Reading package lists...
2022-06-02T22:43:36.8142032Z Building dependency tree...
2022-06-02T22:43:36.8158628Z Reading state information...
2022-06-02T22:43:36.9895304Z dosfstools is already the newest version (4.1-2).
2022-06-02T22:43:36.9896975Z dosfstools set to manually installed.
2022-06-02T22:43:36.9898794Z zip is already the newest version (3.0-11build1).
2022-06-02T22:43:36.9899993Z binfmt-support is already the newest version (2.2.0-2).
[... cut ...]
2022-06-02T22:47:18.3181396Z PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
2022-06-02T22:47:18.3181800Z VERSION_CODENAME=bionic
2022-06-02T22:47:18.3182050Z UBUNTU_CODENAME=bionic
2022-06-02T22:47:18.3216330Z + uname -a
2022-06-02T22:47:18.3560562Z Linux fv-az177-982 5.13.0-1025-azure #29~20.04.1-Ubuntu SMP Thu May 19 14:50:45 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
2022-06-02T22:47:18.3585908Z + ping google.de
2022-06-02T22:47:18.4281143Z ping: google.de: Name or service not known
2022-06-02T22:47:18.4350989Z ##[error]Process completed with exit code 2.

pguyot, 2022-06-06:
After few attempts, I was able to get it right. Could you please test it again?
BTW, you probably want to ping with `-c 1`.

closed 2022-06-06 by pguyot

raphaelscholle, 2022-06-06:
looks great, didn't find any bugs, really big thank you :)

pguyot, 2022-06-07:
Thanks for testing. Just released that as v2.4.0

commit 4b9a26dc (2022-06-06): Move instead of copying resolv.conf, test with ping (fix #38)
commit 4677e2ae (2022-06-06): Move instead of copying resolv.conf, test with nslookup (fix #38)
commit 284a45c9 (2022-06-06): Move instead of copying resolv.conf, test with host (fix #38)
commit 1be4bc9d (2022-06-06): Move instead of copying resolv.conf, test with host (fix #38) (#40)