pguyot/arm-runner-action #44: Having difficulty copying build artefact from image into repository

eggbean, 2022-07-20:
I cannot figure out how to copy a built compiled binary from the image to the cloned repository. I am building statically linked tmux binaries for armv7l and aarch64. This is the relevant part of the action yaml:

```
jobs:
  build_and_upgrade:
    runs-on: ubuntu-latest
    strategy:
      matrix:
        arch: [armv7l, aarch64]
        include:
        - arch: armv7l
          cpu: cortex-a7
          base_image: raspios_lite:latest
        - arch: aarch64
          cpu: cortex-a53
          base_image: raspios_lite_arm64:latest

    steps:
    - uses: actions/checkout@v3
      with:
        token: ${{ secrets.github_token }}

    - name: Upgrade buildscript
      run: wget https://github.com/mjakob-gh/build-static-tmux/raw/master/build-static-tmux.sh -O bin/scripts/build-static-tmux.sh

    - uses: pguyot/arm-runner-action@v2
      with:
        base_image: ${{ matrix.base_image }}
        cpu: ${{ matrix.cpu }}
        cpu_info: ${{ matrix.cpu_info }}
        copy_artifact_path: /tmp/tmux-static/bin/tmux
        image_additional_mb: 3072
        commands: |
            test $(uname -m) = ${{ matrix.arch }}
            /bin/sh bin/scripts/build-static-tmux.sh
            mv /tmp/tmux-static/bin/tmux bin/${{ matrix.arch }}
```

And here is the relevant part of the log:

```
 960   │ 2022-07-20T15:13:29.7141582Z Resulting files:
 961   │ 2022-07-20T15:13:29.7202303Z ----------------
 962   │ 2022-07-20T15:13:29.7202708Z Standard tmux binary:   /tmp/tmux-static/bin/tmux.linux-armv7l.gz
 963   │ 2022-07-20T15:13:29.7203157Z Stripped tmux binary:   /tmp/tmux-static/bin/tmux.linux-armv7l.stripped.gz
 964   │ 2022-07-20T15:13:30.4407740Z
 965   │ 2022-07-20T15:13:30.4408383Z ----------------------------------------
 966   │ 2022-07-20T15:13:30.4739044Z Duration: 0h 59m 45s
 967   │ 2022-07-20T15:13:30.4742236Z
 968   │ 2022-07-20T15:13:30.4745798Z + mv /tmp/tmux-static/bin/tmux bin/armv7l
 969   │ 2022-07-20T15:13:30.5106370Z + rc=0
 970   │ 2022-07-20T15:13:30.5119302Z + '[' -f /home/actions/temp/arm-runner/mnt/tmp/github_env.sh ']'
 971   │ 2022-07-20T15:13:30.5119573Z + exit 0
 972   │ 2022-07-20T15:13:30.5174397Z ##[group]Run case "yes" in
 973   │ 2022-07-20T15:13:30.5174646Z [36;1mcase "yes" in[0m
 974   │ 2022-07-20T15:13:30.5174832Z [36;1myes|true)[0m
 975   │ 2022-07-20T15:13:30.5175028Z [36;1m    set -x[0m
 976   │ 2022-07-20T15:13:30.5175215Z [36;1m;;[0m
 977   │ 2022-07-20T15:13:30.5175379Z [36;1mesac[0m
 978   │ 2022-07-20T15:13:30.5175577Z [36;1mrepository_path=[0m
 979   │ 2022-07-20T15:13:30.5175830Z [36;1mif [ "${repository_path}x" = "x" ]; then[0m
 980   │ 2022-07-20T15:13:30.5176122Z [36;1m    repository_name=`basename /home/runner/work/.dotfiles/.dotfiles`[0m
 981   │ 2022-07-20T15:13:30.5176425Z [36;1m    repository_path=/${repository_name}[0m
 982   │ 2022-07-20T15:13:30.5176640Z [36;1mfi[0m
 983   │ 2022-07-20T15:13:30.5176873Z [36;1mif [ "/tmp/tmux-static/bin/tmuxx" != "x" ] && [ ".x" != "x" ]; then[0m
 984   │ 2022-07-20T15:13:30.5177160Z [36;1m    while read -d\; copy_artifact_path; do[0m
 985   │ 2022-07-20T15:13:30.5177541Z [36;1m        [ -z "${copy_artifact_path}" ] || sudo cp -Rp /home/actions/temp/arm-runner/mnt${repository_path}/${copy_artifact_path} .[0m
 986   │ 2022-07-20T15:13:30.5177902Z [36;1m    done <<< "/tmp/tmux-static/bin/tmux;"[0m
 987   │ 2022-07-20T15:13:30.5178109Z [36;1mfi[0m
 988   │ 2022-07-20T15:13:30.5261294Z shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
 989   │ 2022-07-20T15:13:30.5261557Z ##[endgroup]
 990   │ 2022-07-20T15:13:30.5358142Z + repository_path=
 991   │ 2022-07-20T15:13:30.5358843Z + '[' x = x ']'
 992   │ 2022-07-20T15:13:30.5365885Z ++ basename /home/runner/work/.dotfiles/.dotfiles
 993   │ 2022-07-20T15:13:30.5377222Z + repository_name=.dotfiles
 994   │ 2022-07-20T15:13:30.5380848Z + repository_path=/.dotfiles
 995   │ 2022-07-20T15:13:30.5384578Z + '[' /tmp/tmux-static/bin/tmuxx '!=' x ']'
 996   │ 2022-07-20T15:13:30.5384848Z + '[' .x '!=' x ']'
 997   │ 2022-07-20T15:13:30.5387629Z + read '-d;' copy_artifact_path
 998   │ 2022-07-20T15:13:30.5388706Z + '[' -z /tmp/tmux-static/bin/tmux ']'
 999   │ 2022-07-20T15:13:30.5389353Z + sudo cp -Rp /home/actions/temp/arm-runner/mnt/.dotfiles//tmp/tmux-static/bin/tmux .
1000   │ 2022-07-20T15:13:30.5529130Z cp: cannot stat '/home/actions/temp/arm-runner/mnt/.dotfiles//tmp/tmux-static/bin/tmux': No such file or directory
1001   │ 2022-07-20T15:13:30.5553324Z ##[error]Process completed with exit code 1.
1002   │ 2022-07-20T15:13:30.5701387Z Post job cleanup.
```

The buildscript makes a tmux binary at `/tmp/tmux-static/tmux` which I want to copy to `bin/armv7l/` in the repository, but it cannot find it. What am I doing wrong?

Also, why is this taking over an hour to run? I have tried increasing the memory to up to 4096MB, but nothing changes. Is there anything that I can do to make it run faster or is this to be expecting with GitHub Actions or this action specifically?

Thanks.

pguyot, 2022-07-21:
`copy_artifact_path` is a relative path in the emulated environment. It cannot be absolute as shown by
your log. `copy_artifact_dest` can be absolute and is a path in the "host" environment.

You probably want to swap the values of both parameters.

eggbean, 2022-07-23:
Thanks. I was confused about you saying that I probably want to swap the parameters. I don't know why you thought that, but after experimenting and finally installing `tree` and searching through the whole filesystem, I realised that my variables should be this:

    copy_artifact_path: ../tmp/tmux-static/bin/tmux
    copy_artifact_dest: bin/${{ matrix.arch }}

Everythings is working fine, but I realised that I cannot use a jobs matrix and have the artefacts all get copied to the repository, so I do the builds sequentially. The x86_64 build takes 3 minutes, but the arm build take 62 and 81 mins, so the whole workflow takes 2hrs and 40mins.

This isn't a big deal, but I just want to make sure that this is to be expected or if there might be something wrong that I've done?

pguyot, 2022-07-23:
This is definitely slow. I played with bind mount but it didn't make such a big difference to my slowest actions.

I suggest you find a way to modify your workflow to build the two arm builds in parallel.

closed 2022-07-23 by pguyot