pguyot/arm-runner-action #47: add support for ubuntu raspi

razr, 2022-10-07:
thanks for your great project, I have tried to use it with ubuntu as:
```yaml
jobs:
  build:
    runs-on: ubuntu-22.04
    steps:
    - uses: actions/checkout@v2
    - uses: pguyot/arm-runner-action@v2
      with:
        base_image: https://cdimage.ubuntu.com/releases/22.04.1/release/ubuntu-22.04.1-preinstalled-server-arm64+raspi.img.xz
        commands: |
            uname -a
```
and get

```bash
cp: cannot stat '/home/actions/temp/arm-runner/mnt/etc/resolv.conf': No such file or directory
cp: not writing through dangling symlink '/home/actions/temp/arm-runner/mnt/etc/resolv.conf'
```

I think this is because:

```
ls -la /etc/resolv.conf
/etc/resolv.conf -> ../run/resolvconf/resolv.conf
```

pguyot, 2022-10-08:
This was fixed in 2.4, but the tag v2 was not moved.
It should be fixed now.

razr, 2022-10-08:
thanks for looking into it.
It looks like that

```
    runs-on: ubuntu-22.04
```

makes the difference. I have [two jobs](https://github.com/razr/hello-world-docker-action/blob/main/.github/workflows/rpi.yaml), both are identical, with the difference of where they `runs-on`. Apparently `ubuntu-latest` is `focal`, whereas `ubuntu-22.04` is `jammy`, The job with `ubuntu-latest` [runs just fine](https://github.com/razr/hello-world-docker-action/actions/runs/3211967281/jobs/5250497430), but the job with `ubuntu-22.04` [fails](https://github.com/razr/hello-world-docker-action/actions/runs/3211967281/jobs/5250497467).

pguyot, 2022-10-08:
It seems to be a different error now, not related to resolv.conf, but instead related to `which`, which cannot find /bin/sh within the chroot with this version of ubuntu for the host.

razr, 2022-10-08:
yes, this is weird.

```
+ shell_path=
Error: Process completed with exit code 255.
```

pguyot, 2022-10-09:
Fixed in v2.5.1

closed 2022-10-09 by pguyot

razr, 2022-10-09:
It works on my side, thanks!
There are some warnings at the end
 
```bash
Run sudo rm /usr/bin/qemu-arm-static0
update-binfmts: warning: no executable /usr/bin/qemu-arm-static0 found, but continuing anyway as you request
update-binfmts: warning: current package is <local>, but binary format already installed by arm-runner-action; not removing.
update-binfmts: warning: no executable /usr/bin/qemu-aarch64-static0 found, but continuing anyway as you request
update-binfmts: warning: current package is <local>, but binary format already installed by arm-runner-action; not removing.
update-binfmts: warning: unable to enable binary format qemu-arm: another format with the same magic already exists and this is a fix-binary format: No such file or directory
update-binfmts: warning: unable to enable binary format qemu-arm
update-binfmts: warning: unable to enable binary format qemu-aarch64: another format with the same magic already exists and this is a fix-binary format: No such file or directory
update-binfmts: warning: unable to enable binary format qemu-aarch64
```