pguyot/arm-runner-action #54: Cannot cd into repository directory and issue with ubuntu-18.04

rafal1137, 2022-10-13:
When running your action on `ubuntu-latest` and `ubuntu-18.04` I did encountered such issues.

Part of the github workflow file
```yml
  raspberry_pi-mod:
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2.1.0
        with:
          submodules: true
          fetch-depth: 0

      - uses: pguyot/arm-runner-action@v2
        with:
           image_additional_mb: 4096
           base_image: raspi_4_bullseye:20220121
           cpu: cortex-a53
           cpu_info: cpuinfo/raspberrypi_4b
           bind_mount_repository: /opt/etlegacy
           optimize_image: no
           commands: |
             sudo apt-get update
             sudo apt-get -y install build-essential libudev-dev automake zlib1g libraspberrypi0 \
             libraspberrypi-bin libraspberrypi-dev libx11-dev \
             nasm autoconf git cmake zip gcc g++ libtool ninja-build
             cd /opt/etlegacy
             git fetch --tags --force
             ./easybuild.sh build -RPI -mod -ninja
```

### Ubuntu-latest

```/tmp/commands.sh: 2: cd: can't cd to /etlegacy```

Most of the log showing what happened during

```
Run case cortex-a53 in
Run bash /home/runner/work/_actions/pguyot/arm-runner-action/v2/download_image.sh raspi_4_bullseye:20220121
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Run sudo bash /home/runner/work/_actions/pguyot/arm-runner-action/v2/mount_image.sh /home/runner/work/_temp/arm-runner/arm-runner.img 4096 no 2 1
4096+0 records in
4096+0 records out
4294967296 bytes (4.3 GB, 4.0 GiB) copied, 19.3924 s, 221 MB/s
Created loopback device /dev/loop3
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
RASPIROOT: 19851/102544 files (0.2% non-contiguous), 158239/409600 blocks
resize2fs 1.45.5 (07-Jan-2020)
Resizing the filesystem on /dev/loop3p2 to 1458176 (4k) blocks.
The filesystem on /dev/loop3p2 is now 1458176 (4k) blocks long.

Finished resizing disk image.
/dev/loop3: msdos partitions 1 2
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
mv: cannot stat '/home/actions/temp/arm-runner/mnt/etc/resolv.conf': No such file or directory
Run if [ -e /home/runner/work/etlegacy/etlegacy/cpuinfo/raspberrypi_4b ]; then
Run case "yes" in
+ repository_path=
+ '[' x = x ']'
++ basename /home/runner/work/etlegacy/etlegacy
+ repository_name=etlegacy
+ repository_path=/etlegacy
+ case "yes" in
+ exit_on_fail=e
+ shell_with_opts=/bin/sh
+ shell=/bin/sh
+ '[' /bin/sh '!=' /bin/sh ']'
+ shell_opts=
+ shell_package=
+ '[' -x /home/actions/temp/arm-runner/mnt//bin/sh ']'
++ sudo chroot /home/actions/temp/arm-runner/mnt /bin/sh -c 'command -v /bin/sh'
+ shell_path=/bin/sh
++ dirname /home/actions/temp/arm-runner/mnt/etlegacy
+ sudo mkdir -p /home/actions/temp/arm-runner/mnt
+ case /opt/etlegacy in
+ '[' nox '!=' x -a nox '!=' nox ']'
+ chroot_script_dir=/tmp
+ script_dir=/home/actions/temp/arm-runner/mnt/tmp
+ mkdir -p /home/actions/temp/arm-runner/mnt/tmp
+ script_path=/home/actions/temp/arm-runner/mnt/tmp/commands.sh
+ touch /home/actions/temp/arm-runner/mnt/tmp/commands.sh
+ '[' xe '!=' '' ']'
++ basename /bin/sh
+ shell_basename=sh
+ '[' sh = sh -o sh = bash -o sh = ksh -o '' = ksh ']'
+ echo 'set -xe'
+ case "no" in
+ case "no" in
+ echo 'cd /etlegacy'
+ cat
+ '[' nox '!=' x -a nox '!=' nox ']'
+ sudo chroot /home/actions/temp/arm-runner/mnt /bin/sh /tmp/commands.sh
+ cd /etlegacy
/tmp/commands.sh: 2: cd: can't cd to /etlegacy
Error: Process completed with exit code 2.
Run sudo umount /home/actions/temp/arm-runner/mnt/proc/cpuinfo
Run case /opt/etlegacy in
mv: cannot stat '/home/actions/temp/arm-runner/mnt/etc/_resolv.conf': No such file or directory
Run sudo rm /usr/bin/qemu-arm-static0
update-binfmts: warning: no executable /usr/bin/qemu-arm-static0 found, but continuing anyway as you request
update-binfmts: warning: current package is <local>, but binary format already installed by arm-runner-action; not removing.
update-binfmts: warning: no executable /usr/bin/qemu-aarch64-static0 found, but continuing anyway as you request
update-binfmts: warning: current package is <local>, but binary format already installed by arm-runner-action; not removing.
update-binfmts: warning: unable to enable binary format qemu-arm: another format with the same magic already exists and this is a fix-binary format: No such file or directory
update-binfmts: warning: unable to enable binary format qemu-arm
update-binfmts: warning: unable to enable binary format qemu-aarch64: another format with the same magic already exists and this is a fix-binary format: No such file or directory
update-binfmts: warning: unable to enable binary format qemu-aarch64
```

### Ubuntu-18.04

```
Run case cortex-a53 in
update-binfmts: unrecognized option '--unimport'
Try 'update-binfmts --help' or 'update-binfmts --usage' for more information.
Error: Process completed with exit code 2.
Run sudo umount /proc/cpuinfo
umount: /proc/cpuinfo: not mounted.
Error: Process completed with exit code 32.
Run case /opt/etlegacy in
/home/runner/work/_actions/pguyot/arm-runner-action/v2/cleanup_image.sh: line 6: $3: unbound variable
Error: Process completed with exit code 1.
Run sudo rm /usr/bin/qemu-arm-static0
update-binfmts: warning: no executable /usr/bin/qemu-arm-static0 found, but continuing anyway as you request
update-binfmts: warning: /var/lib/binfmts/arm-runner-action-qemu-arm does not exist; nothing to do!
update-binfmts: warning: no executable /usr/bin/qemu-aarch64-static0 found, but continuing anyway as you request
update-binfmts: warning: /var/lib/binfmts/arm-runner-action-qemu-aarch64 does not exist; nothing to do!
update-binfmts: warning: unable to open /usr/share/binfmts/qemu-arm: No such file or directory
update-binfmts: warning: couldn't find information about 'qemu-arm' to import
update-binfmts: exiting due to previous errors
Error: Process completed with exit code 2.
```

pguyot, 2022-10-13:
`bind_mount_repository` is a boolean. The path should be set with `copy_repository_path`. I agree the name of the option can be misleading.

Could you please try and confirm it works?

```yaml
bind_mount_repository: true
copy_repository_path: /opt/etlegacy
```

rafal1137, 2022-10-13:
@pguyot Yes it does work now.
I am assuming 18.04 is too old to take a look at  ?

pguyot, 2022-10-13:
Isn't ubuntu 18.04 deprecated with GitHub? How long do you mean to use it? The action has just been made recently compatible with ubuntu 22.04, maybe breaking compatibility with 18.04 though.

rafal1137, 2022-10-13:
> Isn't ubuntu 18.04 deprecated with GitHub? How long do you mean to use it? The action has just been made recently compatible with ubuntu 22.04, maybe breaking compatibility with 18.04 though.

Yes it has been marked as depracated. But yeah 18.04 should not be touched anymore.

closed 2022-10-13 by rafal1137