pguyot/arm-runner-action #55: Build x64 image?

chetbox, 2022-10-13:
Is it possible to build a x64 image?

This GitHub Action is great for dealing with all the things one might need to think about to create an ARM system image that's ready to go. We also need to create images (with the same setup!) for Intel x64 machines and it would be great if we could use this action to run with QEMU using `x86_64` so that we're using the same setup.

The README doesn't mention this so I wondered if it's possible as is or with a straightforward modification?

pguyot, 2022-10-13:
Do you have an image to test this with? As long as there is no specific setting to pass to qemu (for cpu), it's mostly about copying qemu-user-static binary for x86_64 to the chroot and reconfiguring binfmt (for ubuntu 22.04 host).

chetbox, 2022-10-14:
The Jammy Server image from https://www.armbian.com/uefi-arm64/ or and Ubuntu cloud image from https://cloud-images.ubuntu.com/minimal/releases/ seem like good test cases although I'm not sure how much UEFI complicates things. 

The official Ubuntu/Debian Server images require a user to go through an installer to set up a user account, etc. so I don't think they are suitable.

pguyot, 2022-10-15:
Sorry I was confused.

arm64 images are aarch64 (it's a synonym) and they are supported.

x64 images are amd64 (like the host) and should also work non-emulated as the action performs a chroot.

pguyot, 2023-02-04:
Closing this as building x64 images should work out of the box. Please reopen if this is not the case, with details on the failure.

closed 2023-02-04 by pguyot

chetbox, 2023-02-07:
Thanks @pguyot. I tried to get this working but got stuck with a problem with a GPT partition table. If I can get a good repro of rhag I'll make an issue/PR.