pguyot/arm-runner-action #59: Some of the Enviroment Variables imported by import_github_env are not set

rafal1137, 2022-11-02:
When I did set `import_github_env` to `true` I did noticed that particular env variable is not set from github workflow.
I am speaking of `CI` that my code depends on.

```cmake
		if("$ENV{CI}" STREQUAL "true")
		/// more code
```
This part of code was never reached. I had to manually set in commands to make it work as expected
`export CI=true`

PS: I had tried setting env variable using the worklow itselt
```yml
env:
    CI: true
```

pguyot, 2022-11-07:
Could you please share your workfile?
There is a test for import_github_env.

Documentation is here: https://github.com/pguyot/arm-runner-action#import_github_env
Test is here: https://github.com/pguyot/arm-runner-action/blob/main/.github/workflows/test-import_github_env.yml

pguyot, 2022-11-12:
Oh, I misunderstood your request.

closed 2022-11-12 by pguyot

pguyot, 2022-11-12:
Fixed in v2.5.2, you don't need import_github_env which serves another purpose.

commit feca5290 (2022-11-12): Pass environment variables to script (fix #59)
commit e04ca3be (2022-11-12): Pass environment variables to script (fix #59)