pguyot/arm-runner-action #60: Arm actions running runs as root and not the runner user

theofficialgman, 2022-11-09:
github actions runs with passwordless sudo as a non-root user.

arm actions runner runs as root (`uid=0(root) gid=0(root) groups=0(root) root`) and makes performing commands not the same as if it were run within the standard ubuntu actions (`uid=1001(runner) gid=121(docker) groups=121(docker),4(adm),101(systemd-journal) runner`).

this should be corrected so that arm-runner-action runs as the `runner` user with passwordless sudo enabled.

pguyot, 2022-11-12:
I'm ok with it, however this is a breaking change and probably should be an option.

theofficialgman, 2022-12-28:
any update on this? I would like if this could be a feature

pguyot, 2023-01-07:
`runner` user usually wouldn't exist on the target image, in your case `raspios_lite:2021-05-07`.

I am adding a new option to specify the user but you will have to make sure it exists on the image. And if you are using passwordless sudo that it has this capability.

pguyot, 2023-01-07:
This supposes the target image has sudo installed and therefore should definitely be an option.
Do you mean to create a PR with an option so this can be integrated in this repository? Or should I merge #66 instead?

theofficialgman, 2023-01-07:
I don't intend on creating a PR based on what I have done.
its up to you if you want to add the additional functionality from my changes in a toggleable/configurable way

pguyot, 2023-01-07:
OK. I'm merging #66 for now as it is creating value and closing this issue.

PR #66 (2023-01-07): New option to define user to run commands in the target environment
Fix #60
closed 2023-01-07 by pguyot