pguyot/arm-runner-action #64: Docker Daemon Not Running

ElioDiNino, 2023-01-06:
I am looking to use this GitHub action to build, test, and deploy the ARM version of one of the @ubcuas projects. [This](https://github.com/ubcuas/ACOM/blob/workflow-docker/.github/workflows/docker.yml) is the workflow file I have right now (see "build-arm"). I was able to get Docker to install but when I try to run something it says that the daemon is not running ([example of a failed job](https://github.com/ubcuas/ACOM/actions/runs/3851414010/jobs/6562556425)). Doing some research online it seems that this is caused by being inside `chroot`. Is there any way to get around this to run Docker commands?

MichaIng, 2026-09-25:
If you do everything in a Docker container anyway, wouldn't it be easier to just setup Docker (and QEMU) on the host runner? There are well-known actions for this. The `arm-runner-action` basically makes using Docker obsolete, by downloading/using a base image with `chroot` or optionally `systemd-nspawn` (container engine) already. Running Docker inside this base image renders this base image obsolete. Get what I mean?

Turn `host runner` > `chroot ARM base image` > `Docker container`
into `host runner` > `Docker container`
or how this action is supposed to be used: `host runner` > `chroot ARM base image`, running your CI/CD there directly without Docker.

ElioDiNino, 2026-09-25:
Yes this issue is not relevant anymore with the availability of native ARM GitHub runners that have been released in the 4 years since writing.

closed 2026-09-25 by ElioDiNino