pguyot/arm-runner-action #69: shell files under profile.d are not sourced

gpongelli, 2023-02-14:
Hi @pguyot , thanks for this useful action.

I'm creating some ARM images with built python and rust inside them, but when I'll use them seems shell files under `/etc/profile.d/` , that contain environment variables with paths of just built languages, are not "sourced", so the execution ends with error.

This is what I do:
```
     - name: Create release image
        uses: pguyot/arm-runner-action@v2.5.2
        id: arm_runner_install
        with:
          base_image: ${{ matrix.base_image }}
          cpu: ${{ matrix.cpu }}
          cpu_info: ${{ matrix.cpu_info }}
          shell: bash
          copy_repository_path: /opt/repo
          image_additional_mb: 4096
          commands: |
            echo "------------------ INSTALL RUST ------------------"
            export CARGO_HOME=$CARGO_PATH
            export RUSTUP_HOME=$RUSTUP_PATH
            curl https://sh.rustup.rs -sSf | bash -s -- -y --no-modify-path
            echo "------------------ INSTALL PYTHON ------------------"
            mkdir -p $PYTHON_INSTALL_PATH 
....
            echo "------------------ CHECK ------------------"
            export PATH=$CARGO_PATH/bin:$PYTHON_INSTALL_PATH/bin:$PIPX_BIN_PATH:$PATH
            export LD_LIBRARY_PATH=$PYTHON_INSTALL_PATH/lib:$LD_LIBRARY_PATH
            source $CARGO_HOME/env
            rustc --version
            python3 --version
            echo "------------------ EXPORT ENV ------------------"
            echo "export PATH=$CARGO_PATH/bin:$PYTHON_INSTALL_PATH/bin:$PIPX_BIN_PATH:$PATH" >> /etc/profile.d/env.sh
            echo "export LD_LIBRARY_PATH=$PYTHON_INSTALL_PATH/lib:$LD_LIBRARY_PATH" >> /etc/profile.d/env.sh
            cat $CARGO_HOME/env >> /etc/profile.d/rust.sh
            echo "export CARGO_HOME=$CARGO_PATH" >> /etc/profile.d/rust.sh
            echo "export RUSTUP_HOME=$RUSTUP_PATH" >> /etc/profile.d/rust.sh
```

[this is the full build script](https://github.com/gpongelli/arm-runner-python/blob/1.2.1b1/.github/workflows/rpi_build.yaml) , and this is [its execution](https://github.com/gpongelli/arm-runner-python/actions/runs/4175083700) that builds correctly all the images to [release](https://github.com/gpongelli/arm-runner-python/releases/tag/1.2.1b1).

As you can see, I do a "CHECK" phase after python and rust were installed, and they correctly print the version.

Then I do export the environment variables into those two `/etc/profile.d` files, that are present when I use that image, so I can easily run the tools due the fact `/etc/profile.d` shell files should be automatically sourced when opening a shell.

When the image is then used [here](https://github.com/gpongelli/test-action/blob/main/.github/workflows/rpi_use.yaml)  (another repository, that uses released images from the first one) I expect that those two shell files should be "sourced", so running 
```
rustc --version
python3 --version
```
should print the version.

This does not happen, and I've to "source" those file manually, to have the above command works.
- [execution with manually added source](https://github.com/gpongelli/test-action/actions/runs/4174918341/jobs/7229113745), from [this file commit](https://github.com/gpongelli/test-action/blob/cc1061d51d476dc0c718c26923d7e33b1c270856/.github/workflows/rpi_use.yaml)
- [execution WITHOUT manual source](https://github.com/gpongelli/test-action/actions/runs/4175522972/jobs/7230554567), from [this file commit](https://github.com/gpongelli/test-action/blob/79da43e6a2287daf3c4a42a9c36a388f0da0da1f/.github/workflows/rpi_use.yaml)

Is it possible to have the files under `/etc/profile.d` automatically sourced when running the image ? Or should I put the `source` command manually on user workflow (creating a strong bind between build workflow and user workflow because the need of knowing those paths where tools were installed) ?

Sorry for this long issue, thanks for all.

gpongelli, 2023-02-14:
as a workaround, I should run `for i in /etc/profile.d/*.sh; do . $i; done` as command to do what /etc/profile does.

pguyot, 2023-02-14:
Have you tried passing `-l` to bash like this:

```
shell: bash -l
```
?

I can look further if this doesn't work.

gpongelli, 2023-02-14:
it works, [commit](https://github.com/gpongelli/test-action/blob/21a6c90d052f8da552652f69d826e3b857a282c8/.github/workflows/rpi_use.yaml) without doing manual source, and [its execution](https://github.com/gpongelli/test-action/actions/runs/4176488982/jobs/7232841262).

I didn't read about that flag, thanks.