pguyot/arm-runner-action #70: rustup no space left error

UrNightmaree, 2023-02-17:
Hello, I have a problem when using rustup. When it download the Rust toolchain (`rustc`, `cargo` etc.), it throws an error about no space left.
```
error: failed to extract package (perhaps you ran out of disk space?): No space left on device (os error 28)
```
The code is just this.
```yml
        with:
          base_image: ${{ matrix.image }}:2022-04-04
          shell: /bin/bash
          bind_mount_repository: true
          copy_repository_path: /opt/repo
          copy_artifact_path: artifacts
          copy_artifact_dest: $HOME/artifacts
          commands: |
            export PATH="/opt/repo/scripts:$PATH"

            # fix $HOME being no file or directory
            mkdir $HOME || printf '\x1b[1m$HOME exist!\x1b[0m\n'

            mkdir -p artifacts
            sudo apt-get update
            sudo apt-get install curl git -qy --no-install-recommends

            if ! ${{ github.event.inputs.skip-installing-deps }}; then
              [[ -e scripts/install-deps-${{ github.event.inputs.build }}.sh ]] &&
                bash scripts/install-deps-${{ github.event.inputs.build }}.sh
            fi

            ARTIFACTS=/opt/repo/artifacts bash scripts/build-${{ github.event.inputs.build }}.sh
```
For full code: https://github.com/UrNightmaree/build-things/blob/main/.github/workflows/build-v2.yml#L34

pguyot, 2023-02-17:
You need to enlarge the image to support the files you are adding.

The option is called `image_additional_mb`

https://github.com/pguyot/arm-runner-action#image_additional_mb

closed 2023-02-17 by pguyot

UrNightmaree, 2023-02-17:
Oh thank you