pguyot/arm-runner-action #78: issue: unexpected input(s) "user"

aceisace, 2023-07-26:
Hi @pguyot . Thank you for your awesome work on this github action.
For my use-case, I am trying to install the latest release of Raspberry Pi OS lite and download one of my repos from Github. So far, this action is doing it's work pretty well. Just a short info, the URL to the latest version of Raspberry Pi OS has changed. Maybe this could be updated?

For the next step, I am trying to create a new user, navigate to the home directory of the new user and clone the repo.
First I tried the user parameter, but this did not work:
```sh
Warning: Unexpected input(s) 'user', valid inputs are ['base_image', 'image_additional_mb', 'bind_mount_repository', 'cpu', 'cpu_info', 'commands', 'copy_artifact_path', 'copy_artifact_dest', 'copy_repository_path', 'optimize_image', 'use_systemd_nspawn', 'systemd_nspawn_options', 'bootpartition', 'rootpartition', 'shell', 'shell_package', 'exit_on_fail', 'debug', 'import_github_env', 'export_github_env']
```

I then tried to add a user with the useradd command, like this:
```sh
sudo useradd -p my-super-secure-password test
```
This works, but I cannot cd into the working directory of the user `test`:
`warning: cannot change directory to /home/test: No such file or directory`

I am testing the same setup on the same OS with an Raspberry Pi Zero W. But this seems to work fine. How can I navigate to the user "test's" home directory to clone a repo?

This is the full section of the code:
```sh
- name: run-tests-on-arm
  uses: pguyot/arm-runner-action@v2
  with:
    # Set the base_image to the desired Raspberry Pi OS version
    base_image: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-lite.img.xz
    image_additional_mb: 1500 # enlarge free space to 1.5 GB
    optimize_image: true
    user: test
    commands: |
      echo $PWD && ls
      sudo useradd -p my-super-secure-password test
      sudo usermod -a -G adm,dialout,cdrom,sudo,audio,video,plugdev,games,users,input,netdev,gpio,i2c,spi test
      su - test
      cd /home/test
```

pguyot, 2023-07-26:
According to [man page](https://manpages.debian.org/jessie/passwd/useradd.8.en.html), `useradd(8)` does not create home directory by default.

You may want to use `-m` option with `useradd`.

If it works differently on your device, you should check the content of `/etc/login.defs` file on the image (for example with `cat /etc/login.defs`) and the device.

aceisace, 2023-07-26:
Thanks for your prompt reply @pguyot !
You are right, I did indeed forget the -m option. As the OS I am using is Raspberry Pi OS, the adduser command is actually preferred over the useradd command, which automatically creates the user and the home directory. I chose the more commonly known useradd command as setting a password is easier this way.

Going back to my query, I used this code for testing:

```sh
user: dummy
  commands: |
    whoami
    cd /home
    echo $PWD && ls
    sudo useradd -m -p <secret-password> dummy
    sudo usermod -a -G adm,dialout,cdrom,sudo,audio,video,plugdev,games,users,input,netdev,gpio,i2c,spi dummy
    su - dummy
    whoami
    cat /etc/login.defs
```

Here I am first checking the current user with `whoami`, which outputs `root`. I then try to create a user `dummy` with the `-m` option and grant some privileges according to the Raspberry Pi site.
When attempting to switch switch to the user `dummy` and thereby confirming this `whoami`, the output is not `dummy`, but `root`, see the output: 

```txt
/home
+ ls
pi # why is the user `pi` created here although Raspberry Pi OS no longer uses a pre-set username and password?
+ sudo useradd -m -p *** dummy
+ sudo usermod -a -G adm,dialout,cdrom,sudo,audio,video,plugdev,games,users,input,netdev,gpio,i2c,spi dummy
+ su - inky
+ whoami
root
```

Attached is the output of `cat /etc/login.defs`:
```sh
#
# /etc/login.defs - Configuration control definitions for the login package.
#
# Three items must be defined:  MAIL_DIR, ENV_SUPATH, and ENV_PATH.
# If unspecified, some arbitrary (and possibly incorrect) value will
# be assumed.  All other items are optional - if not specified then
# the described action or option will be inhibited.
#
# Comment lines (lines beginning with "#") and blank lines are ignored.
#
# Modified for Linux.  --marekm

# REQUIRED for useradd/userdel/usermod
#   Directory where mailboxes reside, _or_ name of file, relative to the
#   home directory.  If you _do_ define MAIL_DIR and MAIL_FILE,
#   MAIL_DIR takes precedence.
#
#   Essentially:
#      - MAIL_DIR defines the location of users mail spool files
#        (for mbox use) by appending the username to MAIL_DIR as defined
#        below.
#      - MAIL_FILE defines the location of the users mail spool files as the
#        fully-qualified filename obtained by prepending the user home
#        directory before $MAIL_FILE
#
# NOTE: This is no more used for setting up users MAIL environment variable
#       which is, starting from shadow 4.0.12-1 in Debian, entirely the
#       job of the pam_mail PAM modules
#       See default PAM configuration files provided for
#       login, su, etc.
#
# This is a temporary situation: setting these variables will soon
# move to /etc/default/useradd and the variables will then be
# no more supported
MAIL_DIR        /var/mail
#MAIL_FILE      .mail
[... 276 lines cut ...]
#ENVIRON_FILE
#NOLOGINS_FILE
#ISSUE_FILE
#PASS_MIN_LEN
#PASS_MAX_LEN
#ULIMIT
#ENV_HZ
#CHFN_AUTH
#CHSH_AUTH
#FAIL_DELAY

################# OBSOLETED #######################
#						  #
# These options are no more handled by shadow.    #
#                                                 #
# Shadow utilities will display a warning if they #
# still appear.                                   #
#                                                 #
###################################################

# CLOSE_SESSIONS
# LOGIN_STRING
# NO_PASSWORD_CONSOLE
# QMAIL_DIR
```

pguyot, 2023-07-26:
In your script, `whoami` is executed after `su` finishes, not as part of `su`.

To execute it as part of `su`, you need to pass it with `-c`.

See for example https://unix.stackexchange.com/questions/1087/su-options-running-command-as-another-user#1088

aceisace, 2023-07-26:
Thanks for your fast reply @pguyot . Isn't there an easier way to do this? For example, if I have 20 commands I want to run as a specific user, using `su -c "some_command" user` would be a little tedious. 
I temporarily fixed the issue by running the commands as sudo and then setting permissions back to the new-user. But having support for this in this module would be great. Although there are no more errors now, I still see the 
unexpected input(s) "user" error.

Would be nice to have this feature working though!

While I cannot help with the user-issue, I can create a PR for the latest Raspberry Pi OS Lite release since the links have changed and simply adding /latest at the end of the URL does no longer work.

pguyot, 2023-08-11:
> While I cannot help with the user-issue, I can create a PR for the latest Raspberry Pi OS Lite release since the links have changed and simply adding /latest at the end of the URL does no longer work.

Please do not hesitate to submit a PR.

Regarding permissions, this is no simple topic. Users are different from “host” and “emulated” environment because the user databases are separate. This can be confusing.

aceisace, 2023-08-28:
Created the PR here: #85 👍

I figured a way to make it work more or less with different user-accounts so now it works as expected, with a few roundabouts 💯 

Thanks for your support!

Closing this issue as it has been resolved

closed 2023-08-28 by aceisace