pguyot/arm-runner-action #79: tips for caching apt packages or modified build

ofTheo, 2023-08-10:
Thank you for this action. 
It works super well and is a real help for our @openframeworks CI. 

We're using it to test our C++ code so we basically  `apt-get install`   libs/packages needed and then run our make command with our code.  

The additional packages installation is taking up the bulk of the time (30 mins or so) and I'd love a way to restore a modified version of one of your images with the additional packages added. Is there a mechanism for doing that? Either via actions/cache or another means?

Many thanks!

pguyot, 2023-08-11:
Thank you for your feedback.
The new support for file scheme allows caching by invoking the action several times.

Please see this test: https://github.com/pguyot/arm-runner-action/blob/main/.github/workflows/test-cache.yml

closed 2023-08-11 by pguyot

ofTheo, 2023-08-11:
Amazing - thanks @pguyot!!

commit 973ce287 (2023-08-11): Add support for file scheme in base_image

This allows caching.
Fix #79