pguyot/arm-runner-action #89: How to update repository files inside a cached image?

tugaysolmaz, 2023-10-25:
Hello.

Thanks for the great action tool, we are planning to use this extensively. However, we've been struggling with a minor issue.

Our workflow consists of two jobs:

1. (Optional) Build an image and install requirements, cache after
2. Use cached image to run tests.

However, second job does not update the repo files according to the latest commit. How can we achieve that? I'm providing a minimal workflow below:

````yml
name: rpi4-armhf

on:
  push:
  pull_request:

jobs:
  build-cache-image:
    runs-on: ubuntu-latest
    if: ${{ false }}
    steps:
      - uses: actions/checkout@v4
        name: Checkout
      - uses: pguyot/arm-runner-action@HEAD
        name: build-image
        id: build_image
        with:
          cpu: cortex-a7
          base_image: https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2022-09-26/2022-09-22-raspios-bullseye-armhf.img.xz
          image_additional_mb: 4096
          commands: |
               ...
      - name: Cache OS image
        uses: actions/cache@v3
        id: rpi4-bullseye-armhf

  unit-tests:
    runs-on: ubuntu-latest
    name: rpi4-armhf
    timeout-minutes: 35
    needs: [ build-cache-image ]
    if: ${{ !cancelled() }}
    steps:
      - name: Checkout
        uses: actions/checkout@v4

      - name: Restore cache
        uses: actions/cache/restore@v3

      - uses: pguyot/arm-runner-action@HEAD
        name: Install studio and test
        with:
          base_image: file://${{ runner.temp }}//rpi4-armhf.img
          cpu: cortex-a7
          image_additional_mb: 4096
          import_github_env: true
          commands: |
            python3 run_tests.py
````

pguyot, 2024-04-29:
Did you find a solution to your issue?
Otherwise, I suggest you bind mount the repo in either run. I guess copy doesn't work in the second run because the directory already exists on the image (?). This could be seen in the logs. If you bind mount in the second run, it would overlay. If you bind mount in the first run, repo files wouldn't end up in the image. You could bind mount in both runs.

closed 2024-05-30 by pguyot

tugaysolmaz, 2024-05-30:
Not yet, but I'll retry and get back to you when I do. Thanks for the suggestions.