pguyot/arm-runner-action #90: qemu-aarch64-static0: unable to find CPU model 'cortex-a76'

wrljet, 2023-12-13:
Is there a way for this to include a newer (or patched) qemu to support cortex-a76, such as found in Raspberry Pi 5?

```
++ sudo chroot /home/actions/temp/arm-runner/mnt /bin/sh -c 'command -v /bin/sh'
qemu-aarch64-static0: unable to find CPU model 'cortex-a76'
```

See: [this qemu patch](https://lore.kernel.org/all/20231126231210.112820-1-gshan@redhat.com/T/)

Thanks for any assistance,
Bill

bvobart, 2024-01-14:
Going by https://github.com/pguyot/arm-runner-action/blob/98b8d5a3228f504a8bd54913ab1e23d70f369c6e/action.yml#L102-L127 it seems that you could input this yourself through the `cpu` parameter for this action, provided that the latest version of QEMU bundled with Ubuntu 22.04 (currently the default image used by this action) supports it.

pguyot, 2024-04-29:
Exactly. This can be achieved by:
- using a container or runner that has an apt package with the qemu patch and installing them, so that:
`apt install -y qemu qemu-user-static`
doesn't overwrite the patched qemu version
- passing cpu parameter.

Question is: what is the use case where exactly emulating the Raspberry Pi 5 CPU makes sense. Is it worth the effort and the longer CI time?

pguyot, 2024-05-30:
Ubuntu 22.04 indeed features a rather old version of qemu.
You can upgrade to 24.04 once PR #111 is merged, and then you will be able to pass cortex-a76 in `cpu` option.

commit 5047e19b (2024-05-30): Support ubuntu 24.04 image

Package qemu no longer exists, but only qemu-user-static was required.
Also add support for RaspberryPi 5 emulation with a specific cpu_info to be
used on ubuntu-22.04 and cortex-a76 to be used on ubuntu-24.04.
Clarified that cpu_info is effectless with qemu 8.2+.
Also allow custom qemu.

Fix #90
commit c7a8be23 (2024-05-30): Support ubuntu 24.04 image

Package qemu no longer exists, but only qemu-user-static was required.
Also add support for RaspberryPi 5 emulation with a specific cpu_info to be
used on ubuntu-22.04 and cortex-a76 to be used on ubuntu-24.04.
Clarified that cpu_info is effectless with qemu 8.2+.
Also allow custom qemu.

Fix #90
closed 2024-05-30 by pguyot