pguyot/arm-runner-action #93: Don't know how to uncompress image

PMLS3, 2024-02-23:
Run bash /home/runner/work/_actions/pguyot/arm-runner-action/v2.5.2/download_image.sh https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.7z
Don't know how to uncompress image  *
Error: Process completed with exit code 1.
Run case true in
umount: /tauri-linux: no mount point specified.
Error: Process completed with exit code 32.

nazevice, 2024-02-28:
The url to this image doesn't exist anymore. The new url is: 
[https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.img.xz](https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.img.xz)
As a workaround you can manually add the new url inside your workflow like this:
- uses: pguyot/arm-runner-action@v2.5.2
        with:
          base_image: https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.img.xz

pguyot, 2024-04-29:
This was fixed by #95

closed 2024-04-29 by pguyot