pguyot/arm-runner-action #94: Error: Process completed with exit code 32.

pompushko, 2024-02-26:
Hello
I get following errors during sample pipeline:

```
mount: /home/actions/temp/arm-runner/mnt/proc/cpuinfo: special device /home/runner/work/_actions/pguyot/arm-runner-action/v2/raspberrypi_zero_w does not exist.
Error: Process completed with exit code 32.
```

Here is my pipeline.
```
name: Test architecture matrix
on: [push, pull_request, workflow_dispatch]
jobs:
  build:
    runs-on: ubuntu-latest
    strategy:
      matrix:
        arch: [armv6l, armv7l]
        include:
        - arch: armv6l
          cpu: arm1176
          base_image: raspios_lite:latest
          cpu_info: raspberrypi_zero_w
        - arch: armv7l
          cpu: cortex-a7
          base_image: raspios_lite:latest
          cpu_info: raspberrypi_zero2_w
    steps:
    - uses: pguyot/arm-runner-action@v2
      with:
        base_image: ${{ matrix.base_image }}
        cpu: ${{ matrix.cpu }}
        cpu_info: ${{ matrix.cpu_info }}
        commands: |
            test `uname -m` = ${{ matrix.arch }}
            grep Model /proc/cpuinfo

```

Could you please help me?
Thank you

pompushko, 2024-04-02:
Fixed, missing cpuinfo folder in the path...

closed 2024-04-02 by pompushko

ThexXTURBOXx, 2024-04-04:
How exactly were you able to solve this issue? I have the same issue in my workflow: https://github.com/ThexXTURBOXx/Raspberry-Pi-DRAM-PUF/actions/runs/8554732030/job/23440655551

pompushko, 2024-04-04:
> I was able to solve this issue just now by replacing `cpu_info: raspberry...` with `cpu_info: cpuinfo/raspberry...` (note the `cpuinfo/`). This guide is deprecated and should probably be updated properly: https://github.com/pguyot/arm-runner-action?tab=readme-ov-file#32-and-64-bits

Yep, thank you, already did. Thats why Im closed the issue :D