pguyot/arm-runner-action #105: Something is broken when build via GitHub Actions?

pompushko, 2024-05-09:
Before last upgrade was fine :D

```
+ read '-d;' copy_artifact_path
+ '[' -z main.dist/ ']'
+ sudo cp -Rp /home/actions/temp/arm-runner/mnt/web/main.dist/ .
+ read '-d;' copy_artifact_path
Run sudo umount /home/actions/temp/arm-runner/mnt/proc/cpuinfo
Run case no in
+ loopdev=/dev/loop3
+ mount=/home/actions/temp/arm-runner/mnt
+ image=/home/runner/work/_temp/arm-runner/arm-runner.img
+ optimize=yes
+ rootpartition=2
+ rm /home/actions/temp/arm-runner/mnt/usr/bin/qemu-arm-static0
+ rm /home/actions/temp/arm-runner/mnt/usr/bin/qemu-arm-static
+ rm /home/actions/temp/arm-runner/mnt/usr/bin/qemu-aarch64-static0
+ rm /home/actions/temp/arm-runner/mnt/usr/bin/qemu-aarch64-static
+ '[' -e /home/actions/temp/arm-runner/mnt/etc/_ld.so.preload ']'
+ mv /home/actions/temp/arm-runner/mnt/etc/_ld.so.preload /home/actions/temp/arm-runner/mnt/etc/ld.so.preload
+ '[' -e /home/actions/temp/arm-runner/mnt/etc/_resolv.conf ']'
+ mv /home/actions/temp/arm-runner/mnt/etc/_resolv.conf /home/actions/temp/arm-runner/mnt/etc/resolv.conf
+ [[ -f /home/actions/temp/arm-runner/mnt/tmp/commands.sh ]]
+ rm /home/actions/temp/arm-runner/mnt/tmp/commands.sh
+ [[ -d /home/actions/temp/arm-runner/mnt ]]
Zero-filling unused blocks on boot filesystem...
+ [[ yesx == \x ]]
+ [[ yesx == \y\e\s\x ]]
+ [[ -d /home/actions/temp/arm-runner/mnt/boot ]]
+ echo 'Zero-filling unused blocks on boot filesystem...'
+ cat /dev/zero
+ true
+ sync
+ rm -f /home/actions/temp/arm-runner/mnt/boot/zero.fill
+ echo 'Zero-filling unused blocks on root filesystem...'
Zero-filling unused blocks on root filesystem...
+ cat /dev/zero
+ true
+ sync
+ rm -f /home/actions/temp/arm-runner/mnt/zero.fill
+ umount /home/actions/temp/arm-runner/mnt/dev/pts
+ umount /home/actions/temp/arm-runner/mnt/dev
umount: /home/actions/temp/arm-runner/mnt/dev: target is busy.
+ true
+ umount /home/actions/temp/arm-runner/mnt/proc
+ umount /home/actions/temp/arm-runner/mnt/sys
+ umount /home/actions/temp/arm-runner/mnt/boot
+ umount /home/actions/temp/arm-runner/mnt
umount: /home/actions/temp/arm-runner/mnt: target is busy.
+ true
+ [[ yesx == \x ]]
+ [[ yesx == \y\e\s\x ]]
Resizing root filesystem to minimal size.
+ rootfs_partnum=2
+ rootdev=/dev/loop3p2
+ echo 'Resizing root filesystem to minimal size.'
+ e2fsck -p -f /dev/loop3p2
e2fsck: Cannot continue, aborting.
/dev/loop3p2 is mounted.
```

pguyot, 2024-05-10:
Previously errors in cleanup script were ignored, so the problem might have been unnoticed.

Something prevents /dev to be unmounted and as a result (?), /mnt is not unmounted either. So we cannot optimize the image. If you disable optimize (optimize: no) it should bypass the issue. Do you have a minimal case you could share?

macarooni-man, 2024-05-12:
> If you disable optimize (optimize: no) it should bypass the issue.

https://github.com/macarooni-man/auto-mcs/actions/runs/9053474178/job/24872244608

disabling optimize doesn't appear to resolve the issue.

pguyot, 2024-05-12:
@macarooni-man thank you for sharing this run.
Option name is `optimize_image` and not `optimize` as mentioned in warning.

https://github.com/macarooni-man/auto-mcs/actions/runs/9053474178/job/24872244608#step:4:1

macarooni-man, 2024-05-12:
> Option name is `optimize_image` and not `optimize` as mentioned in warning.

My bad! To clarify, before adding the option at all I was having the same issue as @pompushko. When disabling the `optimize_image` parameter the build finished succesfully. Thanks for your help! I'll leave the run below to compare:

https://github.com/macarooni-man/auto-mcs/actions/runs/9053554984/job/24872427330

pompushko, 2024-05-12:
But why it starts happening?

pguyot, 2024-05-13:
@pompushko This is a regression introduced in v2.6.1. with PR #104 which was a fix of bug #87. Optimize code was a little bit buggy and failed with no notice, which corrupted generated images. #104 fixes the corruption in one case but most importantly makes the build to fail if clean_image script fails as generated images were wrong in this case. Unfortunately, the clean_image script had more reasons to fail within optimize code, for example when the partition could not be properly unmounted.

Thanks to @macarooni-man repository, I have been able to write a test case using Xvfb and fluxbox, and fix the problem in #106. This will be part of 2.6.2.

Best way to avoid any regression (and any bug fix as well) is to stick to a given version rather than just `@v2`. For example, `@v2.6.2`.

commit 6e49dd54 (2024-05-13): Fix an issue where unmount would fail because of lingering processes

Update cleanup_image script to use fuser and umount --lazy --force in last
resort.

Fixes #105
commit 83414dca (2024-05-13): Fix an issue where unmount would fail because of lingering processes

Update cleanup_image script to use fuser and umount --lazy --force in last
resort.

Fixes #105
closed 2024-05-13 by pguyot