pguyot/arm-runner-action #110: Can not run appimages

theofficialgman, 2024-05-30:
attempting to execute appimages results in errors

https://github.com/theofficialgman/testing/actions/runs/9296927833/workflow

https://github.com/theofficialgman/testing/actions/runs/9296927833/job/25586237361#step:2:3145

```
+ sudo /tmp/MuseScore.AppImage install
/tmp/MuseScore.AppImage: 1: �: not found
/tmp/MuseScore.AppImage: 1: �Q�tdR�td���: not found
/tmp/MuseScore.AppImage: 1: /lib/ld-linux-aarch64.so.1GNU�3p�x�_B+N�B6d: not found
/tmp/MuseScore.AppImage: 1: :fGNU�a: not found
/tmp/MuseScore.AppImage: 1: ELFAI�Z@J@8: not found
/tmp/MuseScore.AppImage: 2: Syntax error: "(" unexpected
Error: Process completed with exit code 2.
```

pguyot, 2024-05-30:
Does it work with a physical Raspberry Pi flashed with https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2024-03-15/2024-03-15-raspios-bookworm-arm64.img.xz ?

theofficialgman, 2024-05-30:
I wouldn't have opened an issue if it didn't work on a real system. Yes it does. This is an issue with the qemu binfmt setup.

pguyot, 2024-05-30:
You mean it's a dup of https://github.com/AppImage/AppImageKit/issues/965 ?

theofficialgman, 2024-05-30:
No, this issue is different and doesn't appear to have the same result. See the log there and here.

pguyot, 2024-05-30:
I confirm the problem is related to the magic bytes of the AppImage and the conflict with qemu.
I was able to run the install command by (a) using the trick mentioned in AppImageKit issue 965 here and (b) installing libfuse2.

See this log:
https://github.com/pguyot/testing/actions/runs/9303975038/job/25607495628

And this workflow file:
https://github.com/pguyot/testing/actions/runs/9303975038/workflow

I'll close the issue here as the problem is upstream, feel free to reopen if deemed necessary.

theofficialgman, 2024-05-30:
Editing the header is not a solution for me. I need this to work on generic scripts where the commands executed cannot be modified for limitations in your binfmt. Please re-open.

closed 2024-05-31 by pguyot