pguyot/arm-runner-action #114: Does not run on ARM64 GitHub runners

chetbox, 2024-06-18:
I set up a Linux ARM64 (Beta) GitHub hosted runner wondering if the build would be faster given QEMU ends up running on the same target architecture. I got the following error:

```
update-binfmts: warning: unable to open /usr/share/binfmts/qemu-arm: No such file or directory
update-binfmts: warning: couldn't find information about 'qemu-arm' to unimport
update-binfmts: exiting due to previous errors
warning: could not unimport qemu-arm
update-binfmts: warning: unable to open /usr/share/binfmts/qemu-aarch64: No such file or directory
update-binfmts: warning: couldn't find information about 'qemu-aarch64' to unimport
update-binfmts: exiting due to previous errors
warning: could not unimport qemu-aarch64
/home/runner/work/_temp/1dd63415-b0f0-46f4-9a92-6292765a5c0f.sh: line 32: /usr/bin/sudo: cannot execute binary file: Exec format error
```

pguyot, 2024-06-18:
Is qemu properly installed on this runner?
Do you mean to test for an arm32 Pi or an arm64 one? In the latter case, virtualization with chroot or docker could prove faster, you would avoid emulation.

chetbox, 2024-06-19:
I am building an arm64 image.

I ensured QEMU is installed by running `apt-get install -y qemu-static` but I still get the same error.

I was under the impression this action uses a `chroot` under the hood so my assumption is that no emulation would occur if running on an arm64 runner. Perhaps I have misunderstood?

pguyot, 2024-06-19:
No emulation should occur but the action modifies the binfmt to run the qemu wrapper to ensure a particular CPU is emulated. This might be why sudo doesn't work here.