pguyot/arm-runner-action #118: Cache doesn't work for us

SommerEngineering, 2024-08-24:
Thank you very much for this GitHub Action. It is really great for building ARM releases. We use it for [AI Studio](https://github.com/MindWorkAI/AI-Studio), a desktop app to use local and cloud LLMs. For Linux ARM, we need this GitHub Action.

Building and using the VM works 😀 . However, this one step in the pipeline takes up to 3 hours: 1 1/2 hours to build the image, then another 1 1/2 hours to use the Rust compiler in the VM to build AI Studio. Therefore, I would like to cache the image until either a new Rust version is released or the dependencies in the Rust code change.

Basically, we have also tried this by following your [cache test](https://github.com/pguyot/arm-runner-action/blob/main/.github/workflows/test-cache.yml):

- Starting from [line 487](https://github.com/MindWorkAI/AI-Studio/blob/ae1710488606acc5f463669349d089caf04947b1/.github/workflows/build-and-release.yml#L487) we try to cache the runner image.
- Starting from [line 506](https://github.com/MindWorkAI/AI-Studio/blob/ae1710488606acc5f463669349d089caf04947b1/.github/workflows/build-and-release.yml#L506) we build the ARM runner image.
- Starting from [line 544](https://github.com/MindWorkAI/AI-Studio/blob/ae1710488606acc5f463669349d089caf04947b1/.github/workflows/build-and-release.yml#L544) we copy the built image to the cache.
- Starting from [line 549](https://github.com/MindWorkAI/AI-Studio/blob/ae1710488606acc5f463669349d089caf04947b1/.github/workflows/build-and-release.yml#L549) we start the VM from the image in order to use the Rust compiler for Linux ARM.

Unfortunately, it does not work. The cache is never built. Unfortunately, I cannot explain it. Do you have any idea what could be causing this? We are very grateful for any help.

pguyot, 2024-08-24:
Thank you for your report. There is an issue in the test_cache, indeed it doesn't cache properly because $RUNNER_TEMP isn't properly handled in parameters of the cache action.

Please see fix here with additional check that cache is indeed hit on second job:
https://github.com/pguyot/arm-runner-action/pull/119/files
https://github.com/pguyot/arm-runner-action/actions/runs/10539480436

closed 2024-08-24 by pguyot

SommerEngineering, 2024-09-03:
Thank you very much @pguyot. With your changes, the cache now also works in our repo.

PR #119 (2024-08-24): Fix text_cache with proper cache invocation
Fix #118