pguyot/arm-runner-action #120: qemu package issues when using this runner on github.

mcmillan03, 2024-10-01:
I inherited a github project with CI/CD configurations setup to build and test the project for the Raspberry Pi.  Recently (in the last few days the platform started failing in the configuration step (before it starts compiling the code) it has started failing with errors related to qemu.

The project was using the following from workflows:

 - uses: pguyot/arm-runner-action@v2.5.2
        with:
          base_image: https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.img.xz
          cpu: cortex-a53

And the error message was:

```
Package qemu is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

```
I tried to switch to the latest runners (2.6.5 and 2.6.4) but I am getting different errors related to qemu. The error when I switched to 2.6.4 was:

```
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u10) ...
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
dpkg: error processing package libc-bin (--configure):
 installed libc-bin package post-installation script subprocess returned error exit status 139
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)
Error: Process completed with exit code 100.
```

This is pretty new to me so looking for direction on how to get something working again.

mcmillan03, 2024-10-07:
Additional information:

This happens when I am using "runs-on: ubuntu-latest" which is now apparently using Ubuntu 24.04.1.  When I use "runs-on: ubuntu-22.04", the workflow completes successfully.  This looks like a bad interaction between qemu and the latest ubuntu image available on github.

I am unsure where to report this but I added a comment here: https://github.com/actions/runner-images/issues/10636

SommerEngineering, 2025-01-04:
Thank you @mcmillan03 for creating the issue. We had the same problem with [AI Studio](https://github.com/MindWorkAI/AI-Studio). Fortunately, we were able to find the solution directly in this issue. Switching from `ubuntu-latest` to `ubuntu-22.04` worked perfectly for us as well.

gatk555, 2025-04-02:
Still happening.  The suggestion to specify 22.04 helps, but there is something else.  If anyone who understands this thing would care to look: https://github.com/gatk555/simavr/actions/runs/14227712547/job/39871108929. Another legatee here!

MichaIng, 2026-09-25:
Is this still an issue? I remember that we saw this in our CI as well (unrelated to `arm-runner-action`), and I narrowed it down to the `ldconfig` calls in the `libc-bin` package's `postinst` script:
```sh
if [ "$1" = "triggered" ] || [ "$1" = "configure" ]; then
  ldconfig -r "$DPKG_ROOT/" || ldconfig --verbose -r "$DPKG_ROOT/"
  exit 0
fi
```
But no idea why it causes a segmentation fault in case of certain host QEMU and guest userland combinations. I did not further look into it, since updating our images, to prevent the `libc-bin` upgrade, worked around it, and in the meantime, we dropped Debian Bullseye support. With Bookworm and later guest Debian versions, we have not seen this anymore.

If someone can provide fresh logs with more recent QEMU and Debian versions, maybe a `strace`, we could open an issue at the QEMU bug tracker.

gatk555, 2026-09-26:
No problems recently.  Reliability has been poor for the last 6 months, but that seems unrelated to this specific package.

MichaIng, 2026-09-26:
Regarding reliability: We are constantly struggling with QEMU-incompatibilities systemd was implementing over the last years, starting to use syscalls QEMU does not emulate etc.

E.g. there is this, which prevents starting any systemd unit with load/import credentials directives in the guest, on hosts from Debian Trixie and Ubuntu Noble on: https://gitlab.com/qemu-project/qemu/-/work_items/1962

Our workaround:
```sh
for i in rootfs/lib/systemd/system/*.service
do
	[[ -f $i ]] || continue
	grep -Eq '^(Import|Load)Credential=' "$i" || continue
	sudo mkdir -p "${i/lib/etc}.d"
	echo -e '[Service]\nImportCredential=\nLoadCredential=' | sudo tee "${i/lib/etc}.d/no-credentials.conf"
done
```
Mostly important when actually booting the image, while for builds/tests with `chroot` or `systemd-nspawn` without `-b` flag, as done here, it pops up rarely.

Depending on the _guest_ systemd version, there is another issue with systemd + QEMU: https://github.com/systemd/systemd/issues/39951

Our workaround is to disable all affected sandboxing directives in affected systemd units `$i`:
```sh
echo -e '[Service]\nPrivateUsers=0\nProtectSystem=0\nProtectHome=0\nPrivateTmp=0\nPrivateDevices=0\nProtectKernelModules=0\nProtectControlGroups=0\nProtectKernelTunables=0\nProtectKernelLogs=0\nReadWritePaths=\nProtectProc=default\nProcSubset=all\nNoExecPaths=\nExecPaths=\nInaccessiblePaths=' | sudo tee "rootfs/etc/systemd/system/$i.service.d/container.conf"
```
Not great, but there is no other known workaround.

When when running 32-bit ARM jobs on 64-bit ARM hosts, this might bite you:
- https://github.com/rust-lang/rust/issues/60605
- https://github.com/llvm/llvm-project/issues/41201

Our workaround, is to enable native CP15 barrier instructions on the _host_:
```sh
sudo sysctl -w 'abi.cp15_barrier=2'
```

Feel free to let me know of any issues you face with DietPi base images. Most likely we either faced them in our CI already, or will do so soon when doing any specific software build/test. If there is something we can enhance in DietPi images, we'll do so, but there are a lot of generic issues with QEMU + systemd and 32-bit guests on 64-bit hosts, which need to be addressed on the host.